techninja/cncserver #60: [Meta] Remaining issues for Scratch Support

techninja, 2015-04-19:
Today I ran a 5hr workshop running two bots on two computers at the same time, using the experimental Scratch support with kids aged 5-15 today and I managed to learn a **_lot_**. Apart from various UI bugs and awkwardness, scratch and the API performed admirably and one student was able to use it to draw a star in a box, with a limited amount of fiddling. Some issues I saw:
* Occasionally some commands seemed to be out of order, or misplaced? A wash after a color get, or far more commonly, a move to a position without doing anything. 90% of runs after a time all mysteriously started with a move to some random position on the draw area, then a continuation of business as usual. 
* Setting the speed seems to move to the center, in some circumstances. Or maybe last known turtle position?
* Some of the block "design" is awkward, though once explained it seems to hold itself well. I'd say the worst part is that maximized on a 1920x1080 display, there's no way to see all the blocks. It would likely be a good idea to, if not cull the less useful blocks, to order them by commonality of use, or perhaps even offer two extensions?
* Hard as it may be to find, "early on" tracked turtle position _**can**_ be offset and wrong. A number of drawings started out fine, then at some point simply seemed to find themselves offset in the X axis. This is a low level CNCserver issue and has led to some literally messy issues with the brush dipping in black then trying to get paint from the paper. This issues seemed to never occur again after a clean red stop button reset and park, or after continued use.

It's good we've kept it in beta preview, but I believe it really is quite close. **Some good news**: Some of these issues are going to be easy fixes with no API changes, though others are going to be tricky to track down, I've come up with a clean plan for automated testing that would show us clearly if any commands are sent in the wrong order, or if are doing things they shouldn't by triggering a command or series of commands, then checking the actual serial data sent to what would be expected. Best thing is this will test much faster and let us know if anything breaks support in the future. OK, mor elater, just wanted to document this before it got too late.

techninja, 2015-04-30:
Timing wise, things have been tough since "the downsizing". Good news is, I have an interview scheduled, and plan on getting in three solid 8-10 hour days just for figuring out RP/CNC server, likely starting the coming Monday. Should be a bit like the last revamp, getting all the tickets in order, and lots of PRs to come.

oskay, 2015-04-30:
All sounds good (except for the downsizing part). ;)