techninja/cncserver #91: Custom config for vertical plotter

alexiscz, 2017-07-13:
Hello guys, 

Do you have any idea if EiBotBoard with this CNC server can support vertical plotter (like this one [http://www.gocupi.com/](http://www.gocupi.com/)). 
Both Axidraw and watercolorbot movements are based on two axes (X and Y), I have checked the EBB (EiBotBoard) Command Set webpage and I have seen that the we are supposed to use the commands XM or SM (which makes sense ;)) 
I find this repository helpful and user friendly 👍  but I am pretty sure that won't be able to use EiBotBoard and this CNC server for my project without flashing the board to make it work in my case. 
Let me know if you have suggestions 🙈 

Thx

EmbeddedMan, 2017-07-13:
The EBB can run a plotter like that no problem. Instead of re-programming
the EBB to take the commands being sent from that project's Pi software,
you may want to just modify the Pi software a little to use the existing
EBB commands (SM/XM/LM, etc.)

*Brian

alexiscz, 2017-07-13:
Hey @EmbeddedMan really fast :)

I had this idea but i came to the following problem (maybe i am wrong) :  From my understanding the plotter i want to make doesn't use the xy positions as watercolorbot will do. In my case both motor will have to move if i want to change the X position (same for Y position). So I will to create a new origin and then translate each coordinate to a new one. 
I have started to search for a mathematic solutions ;) then I have thought that it will be easier to code and to debug to change the way the board control the motor. But again maybe i am wrong.

Alex

EmbeddedMan, 2017-07-13:
Alex,

If you're familiar with embedded development, you may be right - re-working
the EBB code may be easier than trying to figure out a bunch of math on the
Pi side.

That said, the EBB uses relative moves, not absolute. So each SM command
just tells how far (in steps) to move each motor, and how long the move
should take. It has no idea what the overall coordinate system or global
coordinates are.

*Brian

alexiscz, 2017-07-13:
Ahah please find my skills bellow : 
Embedded dev : 💩 
Software dev : 👍 👍 
Thank your for your help I will dig deeper the software's way. I was really opened to receive suggestions for my project.

A

oskay, 2017-07-13:
There's really not much to this. There's a simple transformation to convert your desired XY points into 
extended length of the two strings:
https://www.marginallyclever.com/2012/02/drawbot-overview/

Math like this is _vastly_ easier to do on your desktop computer (or laptop or pi or whatever) than on the embedded device.

Then, send relative movement commands to the EBB using the regular SM command.

alexiscz, 2017-08-03:
Hello @oskay thank your feedbacks this what I am currently trying to do. I am working on two sides : 
CNCserver : I have made my own custom ini for my plotter (I will be glad to share it with you as soon it will be done) 
Robotaint : I am tweaking some functions from Robopaint to convert my points. 

I will keep you updated of the result.

Thanks mates!

oskay, 2017-08-03:
Referencing related discussion ( https://github.com/evil-mad/robopaint/issues/293 ), I'm updating the title of this issue.

alexiscz, 2017-08-06:
Ok guys, it was really easy to add (a dirty) support for V Plotter with the getPosChangeData function. Now I am currently dealing with straight lines aberrations. 
[image]

As soon as I will find the perfect solution for aberration I will send you an update.

alexdmejias, 2018-03-27:
@alexiscz can you update me on your progress/project? I'm super interested in building the same thing