techninja/cncserver #94: 250k Baud

reconbot, 2018-02-06:
As of serialport@6.1.0 we now have custom buadrate support for linux. This will allow buadrates on linux of 250k and maybe even faster. I'd love if people here could give it a try and let me know if it work for them. I'm most curious about all the flavors of linux that run on embedded devices.

EmbeddedMan, 2018-02-06:
reconbot - is this for talking over USB to an EBB?

Or for another board?

reconbot, 2018-02-06:
Yes that is what this is used for. Probably every board that's serial over USB, and looking through the code, it's how robotpaint talks to anything. It's used [here](https://github.com/techninja/cncserver/blob/3f3a24d1e058eb3752ddd38af04bc4b9410f4217/runner/cncserver.runner.js#L165).

EmbeddedMan, 2018-02-06:
So the thing you need to keep in mind is that the baudrate for a CDC-ACM USB connection (which is what the EBB uses) doesn't actually use the baudrate value for anything. You can set the baurdrate to 300 baud, and it will still go as fast as USB will allow. You can set it to 3MBaud, and it will still go as fast as USB allows. There's no concept of 'baudrate' in a CDC-ACM connection. So I'm guessing (based on my experience with Windows) that using different baud rates won't make any difference in the actual speed that bytes get from one end of the connection to the other.

reconbot, 2018-02-06:
That's awesome, I didn't know that existed. We also have a small but fix on osx for a possible crash. But this will be more exciting for people living on a baudrate for sure.

oskay, 2018-09-06:
I'm closing this issue as apparently resolved; please feel free to open it if there's more to discuss.

closed 2018-09-06 by oskay