techninja/cncserver #95: Can't connect to AxiDraw

TheAmazingPT, 2018-07-06:
I made sure, that the AxiDraw is properly connected to my Laptop and has power.

My system:
```
Ubuntu 16.04.4 LTS (Xenial Xerus)

Node: 10.5.0
NPM: 6.1.0
```

```
» cat config.ini
httpPort=4242
httpLocalOnly=true
swapMotors=false
maximumBlockingCallStack=100
showSerial=false
serialPath={auto}
bufferLatencyOffset=30
corsDomain=*
debug=false
botType=watercolorbot
scratchSupport=true
flipZToggleBit=false

[invertAxis]
x=false
y=false

[botOverride]
info="Override bot settings E.G. > [botOverride.eggbot] servo:max = 1234"
```

```
» node cncserver.js --botType=axidraw --serialPath=/dev/ttyACM0
Successfully loaded config for AxiDraw! Initializing...
Starting IPC server, waiting for runner client to start...
RUNNER:Connected to CNCServer!
Finding available serial ports...
Available Serial ports: /dev/ttyACM0, /dev/ttyS4
Using passed serial port "/dev/ttyACM0"...
Attempting to open serial port: "/dev/ttyACM0"...
Serial failed to send data. Error #44
CONNECTSERIAL ERROR!
=======Continuing in SIMULATION MODE!!!============
CNC server API listening on localhost:4242
Sending EBB config...
---=== AxiDraw (simulated) is ready to receive commands ===---
Scratch v2 Programming support ENABLED
```

What does `Error #44` mean?
`/dev/ttyACM0` is definitely the correct serial port.
I followed the instructions in your Readme but i can't connect to the AxiDraw :(

Does anyone have an idea why?

oskay, 2018-07-06:
Is the AxiDraw working otherwise?

skiprox, 2018-11-02:
Hi! I'm having the same issue on a Raspberry Pi running Ubuntu.

My system details:

```
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial
```

And what I get when I try to run `cncserver.js`:

```
$ node cncserver.js --botType=axidraw
Successfully loaded config for AxiDraw! Initializing...
Starting IPC server, waiting for runner client to start...
RUNNER:Connected to CNCServer!
Finding available serial ports...
Available Serial ports: /dev/ttyACM0
Using first detected port: "/dev/ttyACM0"...
Attempting to open serial port: "/dev/ttyACM0"...
Serial failed to send data. Error #44
CONNECTSERIAL ERROR!
=======Continuing in SIMULATION MODE!!!============
CNC server API listening on localhost:4242
Sending EBB config...
---=== AxiDraw (simulated) is ready to receive commands ===---
Scratch v2 Programming support ENABLED
```

I'm using Node `v8.12.0` and Npm `6.4.1`. I've confirmed that the AxiDraw works by running `cncserver.js` on my Mac (Node `v8.9.0` and Npm `5.5.1`).

Anything on what `Error #44` is would be helpful, thank you so much!

skiprox, 2018-11-02:
This issue was fixed for me by running with `sudo`—hope this helps someone else, and apologies for bothering anyone.