techninja/cncserver #96: Possible to query the device for size?

vaporstack, 2018-09-04:
Hello, I've got a program that aims to generate artwork at arbitrary window sizes (life with mixed hiDPI / non-high DPI is fun).  I've got CNCserver running and drawing to it and it's all well and good - I am however calculating the aspect ratio manually based on the device I'm using.  Is it possible to read the device sizes via the API so that I'm not using hardcoded values? I have two axidraws of different sizes.

I quite literally would only need aspect ratio but it seems width/height would make more sense from a user POV

Cheers, thanks for the great work.

oskay, 2018-09-04:
The AxiDraw does  not have an onboard record of which model it is; the same driver board, firmware, and defaults are used for all AxiDraw models.

The latest EBB firmware (v 2.5.5) supports saving a *name* to each AxiDraw in nonvolatile memory, so you could use that to encode the device size. However, that feature is not yet supported in cncserver. You can do it through direct commands to the EBB (using the [ST](http://evil-mad.github.io/EggBot/ebb.html#ST), [QT](http://evil-mad.github.io/EggBot/ebb.html#QT) through the AxiDraw software v2.x, or the AxiDraw CLI or python API (v 2.x).

oskay, 2018-09-04:
Not directly, so far as I can see. However, the `botType` setting can tell you which ini is selected; if you have more than one ini for the different AxiDraw units, that might be straightforward.

vaporstack, 2018-09-04:
ah, that works. I've created axidrawxlx.ini etc. It's still hardcoding values but at least I can select between them at runtime instead of compile time.

oskay, 2018-09-06:
Closing this issue as apparently triaged, since there's no planned method to support this "properly" in the near future. Please feel free to open it again if there's more to discuss.

closed 2018-09-06 by oskay