techninja/cncserver #98: Given new feature set, reimplement "example" interface as official, mostly optional GUI with diagnostics, upload, etc.

techninja, 2019-08-23:
What does this include? Some ideas..
* Enable keyboard/arrow key direct relative adjustment (with modifiers)
* Low level: Home(park), Reset, Zero
* Tools/preview interface
* Status window with user interaction area for manual tool changes, disconnection alerts, status and buffer execution progress, Pause/Unpause
* Advanced load SVG into GUI paper for individual customized selection & printing with path specific fills and traces (???)
* SVG API direct load
* RP style fill/trace settings
* Bot & version information
* Advanced custom serial command & return value
* Diagnostics: Simple shape draw, example SVG draw, draw ruler, full range movement, speed stress test

techninja, 2020-01-28:
As a basic user, I can open the interface and upload an SVG image and print it with default usable settings.

techninja, 2020-01-28:
As an advanced user, I can load a full color SVG as a project and customize its exact position and what fill and stroke style it uses with a preview before sending it to print with an estimate of how long it will take and what implements will be required to print it.

techninja, 2020-01-28:
As an advanced user, I can paste in an SVG path string and place it within the preview for printing, selecting colors and fills for the object.

techninja, 2020-01-28:
As a basic user I can calibrate the height of the implement for loading a pen or brush, before printing, or during printing.

techninja, 2020-01-28:
As a Basic user I can import a raster image like a jpeg or PNG, and have it be reinterpreted in multiple selectable ways with no extra settings adjustments, that can be drawn using the bot in either color or black and white.

ithinkido, 2020-01-29:
I would be very interested in a Telnet/ TCP option to connect to controllers such as GRBL- ESP 32. As this is not always the most reliable it would be nice if it had the option to configure flow control. There is something similar in lasergrbl.

Hybrid-Force, 2021-06-15:
I built a web app that use AxiDraw and CNC Server to write cards.

I have custom made single line svg fonts. With the font, I can leverage the browser for layout and positioning. Then I get the paths of the rendered characters, and translate them into pen movement API requests and batch send them to CNC Server which drive the AxiDraw bot.