techninja/cncserver #99: Moving AxiDraw to the absolute X / Y position

MKmeuniere, 2019-10-23:
I want to control the X / Y pen position of AxiDraw using the Restful API and RoboPaint. What does “Accepts direct X / Y absolute pen positions as percentage of total width / height” in the feature or API description mean? I would like to know in more detail the relationship between the values put into X and Y as PUT by Rest and the length that AxiDraw actually moves. I want to move AxiDraw to the absolute X / Y position this way.

oskay, 2019-10-23:
What part of that is unclear? That is a full and complete specification.

MKmeuniere, 2019-10-23:
I apologize for not understanding, this is a fairly rudimentary question. For example, if you move the AxiDraw nib 1 inch in the x direction and 3 inches in the y direction, what are the x and y values ​​you need to specify in the rest?
I find it very helpful to know these things.

oskay, 2019-10-23:
The values in the rest interface are percentages of the width and height, not inches.

MKmeuniere, 2019-10-23:
So is it difficult to specify and move AxiDraw to the absolute position that I actually want to move using the rest interface? I want to use AxiDraw as an actuator that moves based on the sensor value that can acquire the xy direction attached separately.

oskay, 2019-10-23:
No, that should not be difficult. Calculate the XY position that you need, and give it that.

If you are unable to calculate a percentage, you might prefer the python API, where you can specify inches directly.

MKmeuniere, 2019-10-23:
I have tested it several times. The initial position is such that the carriage is brought to the end and the y-axis rail is pushed forward. I assigned x = 50 and y = 50 using the rest interface. Since it was written that the ratio was specified here, AxiDraw thought that the carriage would stop at the center of each XY axis by this substitution. However, the carriage actually moved 10cm in the x-axis direction and did not move in the y-axis direction.
I have no experience with Python, so I want to use the rest interface whenever possible. However, it may be necessary to prepare the environment in some cases.

oskay, 2019-10-23:
And what value of `botType` do you have configured?

MKmeuniere, 2019-10-23:
Does "value of botType" mean "botType: 'watercolorbot'" specified in cncserver.globalConfigDefaults = {} of cncserver.js? I didn't change the value here from the default.

oskay, 2019-10-23:
You have the device configuration set to a different geometry and size of machine than the one that you're operating. If you leave it configured that way, it will not move in the direction nor distance that you expect, unless you are actually connected to a WaterColorBot instead of an AxiDraw.

MKmeuniere, 2019-10-23:
Okay, so can I change botType to another value to get the expected movement in AxiDraw? Also, it would be very helpful to tell us what the value would be in that case, and whether there are other changes required from dapi data.

oskay, 2019-10-23:
I can certainly tell you that if you continue to use the watercolorbot settings, it will not work properly with an AxiDraw.

I am not familiar with "dapi."

MKmeuniere, 2019-10-23:
I understand that this botType does not function properly.
I'm sorry. "dapi" is a typo. I am using the rest client api with the folder name “cncserver-master”.

oskay, 2019-10-23:
An example of setting `botType` is given on the main project page: https://github.com/techninja/cncserver

If you click where it says `machine_types`, there is a list of options.

MKmeuniere, 2019-10-23:
Since I am not used to it, the information is very helpful! 
If I have a problem after trying it, I would like to ask a question here again.

closed 2019-10-23 by oskay

MKmeuniere, 2019-10-23:
I tried it. 
https://github.com/techninja/cncserver/blob/master/machine_types/axidraw.ini
Is it correct to set cncserver.js for various settings shown in this link?
For example,
[controller]
name = EiBotBoard
manufacturer = SchmalzHaus
vendorId = 0x04d8
productId = 0xfd92
baudRate = 9600
position = relative
ack = OK
Do I need to include values like this in other code in the api?
There is also "Add support for AxiDraw 2.0 to cncserver (and RoboPaint)", but I did not know the item that should be set on the RoboPaint software.

MKmeuniere, 2019-10-23:
I was able to select AxiDraw in the botType setting in RoboPaint software.

oskay, 2019-10-23:
So far as I know, those settings in axidraw.ini will work, whether you select AxiDraw in RoboPaint's GUI or if you select the `axidraw` botType in cncserver directly.

MKmeuniere, 2019-10-25:
I was able to solve the above problems by operating RoboPaint's GUI. It was also possible to move AxiDraw to a specified percentage position. Thank you very much!