techninja/cncserver #135: Project status?

dtgreene, 2023-01-17:
Hi, I noticed the project has not been updated in some time and doesn't seem to install with newer versions of npm.  I tried using Node v18 and npm v8 and kept getting post-install build errors due to the `serialport` dependency.  I eventually [made a clone](https://github.com/dtgreene/ebbserver) of the repo and updated the dependencies to work with Node 18 and was wondering if a pull request would be welcome.  I understand it'd have to be on a fork and I could split it up into smaller sections but wanted to see if the project was still active and accepting PRs.

oskay, 2023-01-17:
@techninja Any thoughts?

techninja, 2023-01-18:
I haven't been able to touch a single personal project in years, and I would imagine that will stay for a few years more. Maybe one weekend I will find time to post an update but this will do for now:

**_This project is functionally abandoned_** unless someone in the community wants to help take over issue management. I put too much work into trying to fix all the problems and make new features for v3 back in 2019-2021 before the twins were born, trying hard to get the v3 release out before that, but they showed up nearly three months early and I haven't been able to make time since. Worse, i've been unable to support or manage anything, and now with twins my time is all but fully spent on just living.

I would be so happy to see this project continue in some way, but I simply can't give it time right now. If anyone wants to be added to a team and help fix some things and get a walkthrough of the project and its intents I would be very happy to do so, the fact that this just languishes here saddens me to no end even though it has _some_ good ideas about things. 😄

dtgreene, 2023-01-18:
I'm certainly available to help if help is needed.  I spent the afternoon just getting all of the dependencies up to date and if there's a backlog of issues, I'd be more than willing to contribute.

techninja, 2023-01-19:
@dtgreene email me at james@tn42.com and let's figure this out 😄

closed 2023-02-01 by dtgreene

techninja, 2023-02-03:
For the community: @dtgreene has joined the team and will be managing the project for the foreseeable future! Check the status for the [2.9.0 maintenance release here](https://github.com/users/techninja/projects/1)