techninja/cncserver #139: Near-diagonal motion fails silently

bryangale, 2023-11-22:
When attempting to make near-diagonal pen movements, the pen fails to move, though the API gives no indication of the failure. I'm observing this in cncserver version 2.9.1, operating an AxiDraw V3, running EBB firmware 2.8.1.

The easiest way to observe this with the AxiDraw V3 is, with the pen in the home position, is to PUT `{x: 50, y: 68.8}` (equivalent to 6000 x steps and 5999 y steps) to the `/pen` endpoint. The pen will fail to move, but the API will return success. Additionally, cncserver will **think** the pen has moved, making it easy to try to move the pen beyond the physical limits of the machine on subsequent movements. The only indication of an error will be `!0 Err: <axis2> step rate < 1.31Hz.` outputted to the console.

Similarly, if the pen is at `{x: 100, y: 0}`, attempting to move to `{x: 50, y: 68.8}` will fail with `!0 Err: <axis1> step rate < 1.31Hz.` outputted to the console.

I **think** I understand why this is happening, and it kinda makes sense. (Have I got it right that each motor moves the pen on orthogonal diagonal axes, rather than xy axes?). But it would be better if the API either returns an error in these cases (either detecting ahead of time that the movement would fail, or checking for this message from the firmware). Or splits the movement into multiple move commands that avoid this issue.

I'll need to implement the latter for my current project, whether that's inside or outside cncserver, so any thoughts on how to approach this would be appreciated.

oskay, 2023-11-22:
Best practice is to check commands for sanity before sending them. (The same type of error can happen on the WaterColorBot as well, though in near-horizontal or near-vertical moves, not near-diagonal.)

In the AxiDraw software, if the command is this close to diagonal, we make it actually diagonal, and keep track of the (very small) position error, so that it is typically corrected for on the subsequent move.

bryangale, 2023-11-22:
> In the AxiDraw software, if the command is this close to diagonal, we make it actually diagonal, and keep track of the (very small) position error, so that it is typically corrected for on the subsequent move.

Ah neat. Is [this](https://github.com/evil-mad/axidraw/blob/bb522e363752f116eb403e395f610606fad507f4/inkscape%20driver/motion.py#L947) the check you mean?

It's possible I'm out of my depth, but I'll try and get a PR up soon for a similar check in cncserver.

oskay, 2023-11-22:
Yes, that's one possible way to implement it. The exact check to make is described at http://evil-mad.github.io/EggBot/ebb.html#SM